mxmxmx/temps_utile- #4: Left Encoder button press sync not working

okyeron, 2016-10-08:
I did some poking around in APP_CLK.ino tonight and ran a quick test.

Left-encoder-button press Sync works OK if I make the following modifications:

```
void sync() {
    clk_cnt_ = 0x0;
    div_cnt_ = 0x0;
    ticks_ = 0x0;
  }
```

```
void CLOCKS_leftButton() {
 
  Clock_channel &selected = clock_channel[clocks_state.selected_channel];

  // sync:
  for (int i = 0; i < NUM_CHANNELS; ++i) 
        clock_channel[i].sync();

  if (selected.get_page() == TEMPO) {
    selected.set_page(PARAMETERS);
    clocks_state.cursor = clocks_state.cursor_state;
    selected.update_enabled_settings(clocks_state.selected_channel);
    clocks_state.cursor.AdjustEnd(selected.num_enabled_settings() - 1); 
    return;
  }
}

```
Previously it looked like ticks_ was not being reset in the sync() function. I'm not sure that's the right place to rest ticks_ in this case.
And then in CLOCKS_leftButton() it would not sync if you had the TEMPO page loaded. (because returned before sync() was called?)

I need to learn how to use Github properly - hope this works OK for the moment.

okyeron, 2017-01-10:
Code update on Jan 9, 2017 looks to have fixed this.

closed 2017-01-10 by okyeron

mxmxmx, 2017-01-10:
yeah, it should have the same effect. zeroing ticks_ in the sync() function would be just as fine, or slightly more straightforward. as for the leftbutton behaviour, i've commented out the menu stuff entirely, so the tempo display doesn't toggle when syncing the channels.