mxmxmx/temps_utile- #7: Multiplys continue to pulse after trig is removed

okyeron, 2016-11-07:
Using both TR inputs. One channel set to "-" another set to 2x. If I remove the trig input on TR2 (pull the cable out of the jack) - that channel (which has a mult "2x") continues to send pulses.

Does not happen with "-" or divs "/x"

Also this seems to happen the other way where it's not sending pulses until you change something - haven't been able to replicate this yet.

mxmxmx, 2016-11-07:
hey, 

that's because there's no way knowing the jack has been removed. in theory, some sort of detection mechanism could have been implemented (in parts, in hardware), but i didn't. so when multiplying/running off the internal timer, the module keeps on multiplying; cf 4ms SCM, same thing.

closed 2016-11-07 by mxmxmx

mxmxmx, 2016-11-08:
sure. and sort of, one could do it in software only. ie something like this: if the input trigger frequency increased by this or that amount relative to the previous frequency = someone removed the cable. you'd still get a bunch of triggers at the old frequency. the issue is that in multiplication mode the external clock signals are used to determine the overflow value of an internal timer, they don't actually trigger something; ie it's the last and second-to-last clock that determine the output frequency at any given moment, so when you plug the cable, the timer will sit there.

okyeron, 2016-11-08:
Ah... I see. Great explanation.