mxmxmx/temps_utile- #16: Feature request: persist settings when powered off?

timcowlishaw, 2017-02-06:
Hey there!

Really love the module, however, i've noticed that the configuration of each clock channel doesn't persist through a power cycle, which is problematic for me (and I imagine for others) who set up a patch then transport their system for gigs and can't keep it powered in the meantime. 

Does the hardware support some kind of persistent configuration memory? if so, this'd be a really great feature in future firmware versions, and I'd be happy to contribute to a PR to make that happen!

thanks!

Tim

mxmxmx, 2017-02-06:
hi tim

there's no auto-saving; have you tried https://github.com/mxmxmx/temps_utile-/wiki/using-temps_utile#editing-channelsbasics (last item?)

timcowlishaw, 2017-02-06:
Hah! apologies - that'll teach me to jump straight in without reading the manual. Many thanks - it's a really brilliant module :-)

closed 2017-02-06 by timcowlishaw