mxmxmx/temps_utile- #32: Mute - Trigger option

pppier, 2017-11-17:
hi. thanx for your wonderful work.
it's about TU mute settings and cv signals generated from midi start and stop signal.
can the possibility of driving the mute option using a trigger be interesting, in Temps? 
(i think some midi module, (like befaco midi thing?) give a trigger as a start / stop signal, not a contiue value as doepfer do) :-)

GitBruno, 2018-01-14:
That would be cool. As a workaround today, you can trigger the DAC sequencer as a start / stop signal (High, Low)  to mute another channel.

dmschulman, 2024-02-04:
Pretty sure development has ceased with the TU, but would also enjoy having a way to start/stop/reset the TU sequencer so it can follow an external sequencer and keep the entire rack in sync with that external source.

Banshee700, 2024-02-06:
Hi David,

we are developing the software for the O_C or T_U but both are community
projects
So the code is open source and everyone can build upon it.
Then again..
If you set Output 4 to DAC and the DAC to SEQ pus the clock to TR1 and
reset to RST2 isn't that exactly what you are asking for
You feed a clock signal into TR1 to drive it forward and a second signal to
TR2 to reset it.
If the first clock is absent it will just stop.

best
Sascha