mxmxmx/temps_utile- #47: Inconsistent offset in mult/divide

joodi3k, 2019-03-07:
A channel's behavior is inconsistent when adding offset to divide. I've tested it with no global divide and under 48ppqn. Please see the attached videos as the behavior is so erratic that it defies easy description.

The first video is with no global divide, clock is being provided by a metropolis. 
The second is the same but I'm adjusting offset on channel one instead of two.
The third is with 48ppqn being provided by a beat step pro.

https://www.dropbox.com/s/hysvjxpbmwnbui0/IMG_2467.MOV?dl=0

https://www.dropbox.com/s/menuacn9d1vdb5w/IMG_2478.MOV?dl=0

https://www.dropbox.com/s/jzkq2lrqfmxez9w/IMG_2479.MOV?dl=0

In all three cases the metropolis is providing reset every 4 bars.

mxmxmx, 2019-03-07:
so in the absence of easy description, i'm assuming you're saying that:

1. this issue is about the latest firmware version, 1.3beta: https://github.com/mxmxmx/temps_utile-/tree/1.3b

2. i'll be able to reproduce a/the problem simply by doing the following:
    - set divisor to some value > 1
    - set phase parameter to some value > 0

3. the reset stuff has nothing to do with it?

joodi3k, 2019-03-07:
Actually I’m on v1.2.3. I’ll update and see. My understanding of 1.3b was that it does address divider issues, but not the one I’m experiencing.

To your second item, maybe. Some offset values appear to work. Others change the divisor. Some seem to be the incorrect offset. And still others cause the channel to freeze. None of these inconsistencies are terribly “consistent” in that they don’t seem to always come into play at particular offsets. Except at 50%. That seems to always lock up. Even resyncing the TU doesn’t help.

Reset has nothing to do with it in that the reset does not appear to fix any of the errant behavior. Which is to say if reset caused everything to work as intended again, I’d be fine.

While we’re on the subject, I see a couple of other wrong-seeming behaviors in my 1u TU. In LFSR mode, probability doesn’t change the chance of bits flipping unless I have taps established. And even then it’s kind of hard to tell. Haven’t delved into this further because I keep running into the divider issue.

Also, the save and load behavior does not reflect what’s in the manual. I appear to only have one save slot that loads on power up. Which is really all I need, but now that we’re in the weeds it does seem worth noting.

Thanks

joodi3k, 2019-03-08:
Duh. Latest firmware seems to have cleared up the clock issue, at least with basic testing of the 50% offset. Load/save menu looks correct now. Not sure which version Shay at Plumb loaded mine with when he built it. Or if it just was borked when it got flashed. Thanks for the help.

Jay Reynolds
Tour Manager
Asleep At The Wheel
513-348-7712
jreynolds.bluesky@gmail.com<mailto:jreynolds.bluesky@gmail.com>

mxmxmx, 2019-03-09:
ok, let me know if/when i can close this issue.

re manual: that was updated to reflect v1.3, so you wouldn't have seen the new load/save menu (with 4 slots, not just one)

joodi3k, 2019-03-09:
You can close it. Thanks again.

mxmxmx, 2019-03-10:
perfect, thanks!

closed 2019-03-10 by mxmxmx