mxmxmx/temps_utile- #51: Pulsewidth and Global division setting no longer saved in v1.3

7dials, 2020-01-04:
Hi, I recently tried updating to firmware version 1.3 but have had to revert to version 1.2.3 as I find the new version unusable due to the following issues.

- The pulsewidth set in each channel is no longer saved.
- The global division setting for input TR1 is no longer saved.

This meant I had to reapply these setting every time I turned my modular back on and while the idea of user slots is attractive these issues make the unit unusable in a live situation.

patrickdowling, 2020-04-25:
FWIW this came up in the thread on MW last week, I've put tentative fixes here: [fix/misc_fix](https://github.com/patrickdowling/temps_utile-/tree/fix/misc_fix)

It only later occurred to me to check the issues here.
That's based off master, I'm not really current on which branch might be more up to date.

SimonKunz, 2020-11-19:
@patrickdowling Thank you for publishing your fixes. Just one question: will your fixed version of the firmware keep my calibration settings on the t_u when I install it like described [here](https://github.com/mxmxmx/temps_utile-/wiki/firmware#compile-the-firmware-yourself)?

Thank you!

patrickdowling, 2020-11-19:
Yes, calibration should remain intact.
The preset slots will be lost though -- the versioning isn't particularly smart.

SimonKunz, 2020-11-19:
I don't even use the preset slots to be honest. Thank you :-)