mxmxmx/temps_utile- #57: Can't compile

RobAnt, 2023-06-23:
Hello-

I just built this from a kit from synthCube and am stuck at the firmware portion.

I already had the Arduino IDE as I've used it in the past.  added the Teensy board, and grabbed the code.

When running a verify I get the following error:

```In file included from C:\Users\roban\Documents\Arduino\temps_utile--master\soft\t_u_REV\APP_CLK.ino:29:
C:\Users\roban\Documents\Arduino\temps_utile--master\soft\t_u_REV\TU_pattern_edit.h: In member function 'void TU::PatternEditor<Owner>::handleButtonUp(const UI::Event&)':
C:\Users\roban\Documents\Arduino\temps_utile--master\soft\t_u_REV\TU_pattern_edit.h:407:24: warning: '~' on an expression of type 'bool' [-Wbool-operation]
  407 |       fine_coarse_ = (~fine_coarse_) & 1u;
      |                        ^~~~~~~~~~~~
C:\Users\roban\Documents\Arduino\temps_utile--master\soft\t_u_REV\TU_pattern_edit.h:407:24: note: did you mean to use logical not ('!')?
C:\Users\roban\Documents\Arduino\temps_utile--master\soft\t_u_REV\TU_pattern_edit.h: In instantiation of 'void TU::PatternEditor<Owner>::handleButtonUp(const UI::Event&) [with Owner = Clock_channel]':
C:\Users\roban\Documents\Arduino\temps_utile--master\soft\t_u_REV\TU_pattern_edit.h:293:9:   required from 'void TU::PatternEditor<Owner>::HandleButtonEvent(const UI::Event&) [with Owner = Clock_channel]'
C:\Users\roban\Documents\Arduino\temps_utile--master\soft\t_u_REV\APP_CLK.ino:2703:50:   required from here
C:\Users\roban\Documents\Arduino\temps_utile--master\soft\t_u_REV\TU_pattern_edit.h:407:24: warning: '~' on an expression of type 'bool' [-Wbool-operation]
C:\Users\roban\Documents\Arduino\temps_utile--master\soft\t_u_REV\TU_pattern_edit.h:407:24: note: did you mean to use logical not ('!')?
c:/users/roban/appdata/local/arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld.exe: c:/users/roban/appdata/local/arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/thumb/v7e-m/nofp\libc.a(libc_a-writer.o): in function `_write_r':
(.text._write_r+0x14): undefined reference to `_write'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

I know the tilde is a bitwise NOT and fine_course_ is declared as a bool.  Should I just change the ~ to a ! and hope it works?  Didn't want to mess things up hence asking the question...

Thanks!

patrickdowling, 2023-06-24:
The warning is annoying but not critical., the error
```
...undefined reference to `_write'
```
is from the linker so that's the Teensy/Arduino "stuff". Not sure if there's an easy workaround (maybe turning off lto?) so you might need to use an older version. I gave up on the IDE long ago for platformio -- see [this branch](https://github.com/patrickdowling/temps_utile-/tree/dev/build) -- but really there isn't any on-going maintenance there either :)

booleanvalue, 2024-03-03:
This helps [https://forum.pjrc.com/index.php?threads/undefined-reference-to-_write.71420/](https://forum.pjrc.com/index.php?threads/undefined-reference-to-_write.71420/)

Do note that it still won't compile for 3.2 if the program has no usage of any Print class members without similar:
`void setup() {
  Serial.begin(9600);
  //Serial.println("setup begin"); // won't compile on Teensy 3.x without this line
  printf("Hello World\n");
}`

chailight, 2024-10-08:
I was able to get it to compile using VS Code with the PlatformIO extension.
I had to add the following line to the platformio.ini file

build_flags = -D USB_MIDI -llibc -lc

I'm not 100% sure about the USB_MIDI flag, but that is what is suggested in the following post:
https://forum.pjrc.com/index.php?threads/tutorial-how-to-use-platformio-visual-code-studio-for-teensy.66674/

The suggestion for -llibc -lc comes from here:
https://stackoverflow.com/questions/55784235/platformio-linker-error-when-including-functional

jmsiener, 2024-10-08:
> I was able to get it to compile using VS Code with the PlatformIO extension. I had to add the following line to the platformio.ini file
>
> build_flags = -D USB_MIDI -llibc -lc
>
> I'm not 100% sure about [...]

Have tried marking the explicit versions of Arduino and teensyduino in your platform.ini file? This is going to use a much older Arduino/Teensyduino version but I also can’t tell you what version of either was used to write this code.